MiSTle-Dev/FPGA-Companion #7: RP2040 build instructions - pico_sdk 2.0 compatibility

SanguineBrah, 2024-08-29:
It looks like the latest pico_sdk (2.0) omits stdlib.h so the current won't make without installing the previous release.

harbaum, 2024-08-30:
I would be very surprised if stdlib was not supported in the pico sdk, anymore. Many projects would run into this.

SanguineBrah, 2024-08-30:
It looks like it has moved [here](https://github.com/raspberrypi/pico-sdk/blob/master/src/common/pico_stdlib_headers/include/pico/stdlib.h) in the src folder but set_sys_clock_khz has been removed from the file, which prevents the build.

darius600, 2024-08-31:
I just successfully built FPGA-Companion 1.0.0 with pico-sdk SDK 2.0.0. 
Initially, I encountered some issues. 
They were resolved after I deleted FPGA-Companion and started from scratch by cloning the repository again.

Note the line: _You should delete/recreate all build directories when upgrading from previous versions of the Raspberry Pi Pico SDK 2.0.0_ in the [release notes](https://github.com/raspberrypi/pico-sdk/releases/tag/2.0.0).

Maybe this is the reason here.

SanguineBrah, 2024-08-31:
> I just successfully built FPGA-Companion 1.0.0 with pico-sdk SDK 2.0.0. Initially, I encountered some issues. They were resolved after I deleted FPGA-Companion and started from scratch by cloning [...]

I just did the same. Wiped my directories and started completely from scratch and it worked this time using the 2.0 SDK.

closed 2024-08-31 by SanguineBrah