MiSTle-Dev/FPGA-Companion #15: 8bitdo c64 themed keyboard not recognized

camplusplus, 2024-10-07:
Lsusb shows ID 2dc8:5201 8bitDo Retro Keyboard Receiver. 

I am testing with Commodore 64 core.

harbaum, 2024-10-07:
You can connect a USB uart to the MCU and this will get you some debug output which in turn might give a hint what's happening.

camplusplus, 2025-02-01:
Hello Till,
I connected UART from M0S dock (3 un-soldered pins at USB side)  to M5Cardputer with https://github.com/geo-tp/MicroCOM firmware, but at all baud rates, when I turn on power, I have garbage on screen. 

I used settings:

data bits=8
parity=none
stop bit=1
flow control=no
inverted=no

Can you please tell me if I need to change settings or to use different approach.

Best Regards,
Sasa

harbaum, 2025-02-01:
The bitrate is 2000000 bit/s. 

What USB/uart do you use?

camplusplus, 2025-02-01:
Hi,
UART is directly connected to the M5Cardputer, no USB.

camplusplus, 2025-02-02:
Also no luck with flipper zero as USB UART bridge, output is empty on higher baud rates. I think someone with 
more experience should take over this report.

camplusplus, 2025-02-11:
I bought USB Logic Analyzer and been able to capture data with keyboard and joystick in hub and just 8ditdo directly connected to USB. 
I will try to decode it but maybe you can beat me to it.

[capture-8chanels2MHZ.zip](https://github.com/user-attachments/files/18758807/capture-8chanels2MHZ.zip)

camplusplus, 2025-02-11:
Here it is in rough export
[2.txt](https://github.com/user-attachments/files/18759249/2.txt)

camplusplus, 2025-02-11:
Here it is in rough export

[2.txt](https://github.com/user-attachments/files/18759312/2.txt)

camplusplus, 2025-03-08:
I will try with RP2040-Zero

camplusplus, 2025-03-12:
Flashed and connected Waveshare RP2040-Zero but no luck with any keyboard now 12 red flashes :(

harbaum, 2025-03-13:
What do you mean by "12 red flashes"?

camplusplus, 2025-03-13:
I compiled uf2 for Waveshare device using cmake option and holding boot I added it to pico disk drive. Connected wires and after power on to the fpga I got 12 red blinking led flashes.

camplusplus, 2025-03-13:
My mistake Waveshare RP2040-Zero starts from pin 0 and the print is bad :)
Still no luck connecting it to RP2040.

camplusplus, 2025-07-20:
Hello Till,
Just a info: https://github.com/No0ne/ps2x2pico based on https://github.com/hathach/tinyusb is working with this keyboard. 

Is there a way to use this library instead of CherryUSB, at least or RP2040?