MiSTle-Dev/FPGA-Companion #36: Companion not blinking on Raspberry Pi Pico W

kragil, 2025-01-21:
Hello, I have build the companion for the 2022 Raspberry Pi Pico W strictly by the numbers, but I don't get a blinking LED and the companion does not seem to work. I don't get access to the SD-Card. 

I tried building the .uf2 two times, I don't get errors.

How would I troubleshoot this issue?

Thanks. 

PS. I love this project!

harbaum, 2025-01-21:
Does the release-uf2 work for you?

dijitalxyz, 2025-01-21:
I confirm that the latest compiled pi pico file works with normal pico and W

harbaum, 2025-01-21:
So the question is why @kragil's own build doesn't work.

This e.g. does debug output on the uart Rx/TX pins. But you need a USB/uart adapter to use that.

kragil, 2025-01-22:
Thanks guys! Something is really strange, my compiled versions don't work (don't blink and will not read from SD) AND the release-uf2 will also not work. I've had enough will probably get another pico or even better a M0S, that is the best, right?

harbaum, 2025-01-30:
It would be nice if this could be debugged as other people may run into the same issue.

kragil, 2025-01-30:
Hello, I stopped using the Pi Pico W. I got a cheap noname small Pico with USB-C from China. That one I could flash and the LED blinks like 12 or 13 times and then stops, so it looks OK at first glance. But so far when I connect the new Pico to the Tang Nano it just blinks and does not stop and it won't load the df0.adf from the SD-Card. I will try another SD-Card, but maybe I will just get a M0S. Kinda bummed that I wasn't able to get my "Amiga" working so far ... but probably my own fault somehow.

darius600, 2025-02-18:
I am using a [Raspberry Pi Debug Probe](https://www.raspberrypi.com/documentation/microcontrollers/debug-probe.html) via UART  and connect via Tera Term on Windows. (HTerm and Putty did no work for me.)

I really can recommend this setup since it provides clear debug messages over serial, making it easy to see what the Raspberry Pi is doing.

Maybe there are better ways for reading the serial, but this is the setup I am currently using.

darius600, 2025-02-18:
@kragil which release uf2 are you using - not all of them are blinking afaik.

luciust, 2025-04-24:
> Thanks guys! Something is really strange, my compiled versions don't work (don't blink and will not read from SD) AND the release-uf2 will also not work. I tested my Pico 1 W with another programm and it works just fine. I've had enough, I will probably get another pico or even better a M0S, that is the best, right?

I'm getting similar results, altough only Pi Pico (not the W version) does flash it's LED. But it seems to be related to copying over the u2f file...

harbaum, 2025-04-24:
You can connect a USB uart to the gpio0 and gpio1 and get some debug output. It's at 921600 but/s.

luciust, 2025-04-24:
I did use just GPIO0 (as with some of the sensors) and I'm getting output from Pi Pico W while just powered, now I need to repeat while connected to the shield.