MiSTle-Dev/FPGA-Companion #38: No keyboard for NanoMig with a Raspberry Pi Pico

darius600, 2025-02-13:
I'm unable to get my keyboards to work on NanoMig when using FPGA-Companion with a Raspberry Pi Pico.

**Keyboards tested:**
- Logitech K600
- Rii i4 Mini
- (Miscellaneous others)

Logitech K600 and Rii i4 Mini **work fine with M0S/BL616**, but they fail to function with Raspberry Pi Pico

**Setup:**
- FPGA-Companion Release Build 1.1 (fpga_companion_pico_and_pico-w.uf2)
- Raspberry Pi Pico H (original, SC0917)
- Capturing output via UART using Raspberry Pi Debug Probe

**Logs:**
Attached logs for Logitech K600 and Rii i4 Mini showing output captured via UART.
[rii-i4-relusb.txt](https://github.com/user-attachments/files/18789822/rii-i4-relusb.txt)
[logi-k600-relusb.txt](https://github.com/user-attachments/files/18789823/logi-k600-relusb.txt)

**Additional Information:**
I can compile FPGA-Companion to gather more debug information.

ossipoff, 2025-02-18:
I am using the RP204 Zero and have found that it is more than a little picky about mice and keyboards. I guess that it is something about how the device implement HID. As examples of what works are this mouse https://sordan.ie/product/1494/tank-mouse-white-beige-usb-wireless-bluetooth-amiga-pc--adapter/
And this keyboard https://eu.perixx.com/products/periboard-106

darius600, 2025-02-18:
Thank you, @ossipoff, for the list of working devices.

In the meantime my Logitech K600, Cherry RS 6000, and Logitech 915 TKL are working.
However, I still can't get the Rii i4 Mini to work. 

Based on the logs, the Rii appears to be detected only as a mouse. 
My current guess is that this is caused by TinyUSB, as `tuh_hid_mount_cb ` is called only once for the Rii i4.,

I have no idea why none of the keyboards worked at first. After I switched to a different Raspberry Pi Pico, they started functioning. Oddly enough, after switching back to the original Pico, they (all but the Rii) worked as well. It might be related to the power source I was using but I can’t reproduce it.

closed 2025-02-18 by darius600