MiSTle-Dev/FPGA-Companion #52: Request - join 2 pins to stop Green Light flashing

Higgy69, 2025-04-04:
Hi,
Once a working system has been built and tested, it would be nice if the Green Flashing light could be turned off. I am using a Waveshare RP2040.
I thought this could be achieved by linking 2 unused pins together on the RP2040 PCB and the code reads to see if it needs to toggle light on or off. 

The issue is the constantly blinking light is distracting, it shines through the 3D print case.
The light is very useful when building/diagnosing issues, so I would not suggest removing it altogether and I don't want to try and physically remove the LED.

Thanks

harbaum, 2025-04-04:
I think polling a pin for that purpose really is overkill. May I suggest a drop of black ink over the led? Or just use a pair of pliers and rip it off the board.

The blinking led really serves a purpose: you can see that the CPU has booted, runs the correct firmware and didn't crash yet.

harbaum, 2025-04-22:
I just commited a change that will allow you to specify this in the ini file:

```
; led state (0=blink, 1=on, 2=off)
option led=0

```