MiSTle-Dev/FPGA-Companion #57: Filesystems for the SD-Card

Mangerio, 2025-04-21:
In the description of the **FPGA Companion SPI protocol**, FAT32 and ExFAT are mentioned as filesystems for the SD-Card. 
With FAT32 the size of HDF files is limited to 4GB.
An ExFAT formatted card did not work for the NanoMig...
Isn't it supported?

harbaum, 2025-04-21:
There are now so many variants, that not everything is tested. Exfat is known to work on the bl616. But I am not sure I ever tested this on the rp2040. Is exfat supported by the  pico-sdk? Maybe it has to be enabled explicitly? I don't know ...

Mangerio, 2025-04-22:
Thank you for your reply! I am using the BL616. 
By studying the sourcecodes i found out that a sector size oft at least 512K is required. When formatting the card with the Windows tool a sector size of 32K was preselected by Windows. That did not work. After reformatting with 512K the ExFat filesystem works fine with the MiniMig! I was able to boot from a 8GB HDF. Problem solved!

closed 2025-04-22 by Mangerio