MiSTle-Dev/FPGA-Companion #62: Competition Pro Extra USB Joystick + BL616 + NanoMig

Mangerio, 2025-04-26:
In the Wiki i found the following information:

Proven USB Joysticks
COMPETITION-PRO-EXTRA-USB 	
BL616 (M0S) & Pi Pico
OK

Have tested the Joystick, got no response...
How can i get it to work?
Can i use 2 USB Joysticks?

Thank you!

harbaum, 2025-04-26:
I can test it again. It has been some time since I tested last time. And I am not sure if these competition pro like things are all the same.

Anyway, I also forgot if the Nanomig copes with a second USB joystick. I need to verify that as well.

Mangerio, 2025-04-26:
Thank you very much for your reply! It would be perfect if the NanoMig could handle two USB Joysticks! Me and my 8 year old niece are looking forward to play games together!

I remember that the first generation of the Competiton Pro USB Joystick had a very low polling rate, like 12 times a second and was almost unusable for action games. The second generation solved that problem and got the EXTRA in the name. 

I have the feeling you did not like the idea of two DB9 ports for the NanoMig, but i still hope it will work one day in the future...

Right here on top of the FPGA-Companion Github page is a menu item called "Wiki"