MiSTle-Dev/FPGA-Companion #73: Clarification on internal BL616 builds

HeathenUK, 2025-07-14:
Hi @vossstef, apologies I couldn't raise this on your fork.

I wanted to clarify two things I may be misunderstanding about your most recent commits and builds adding support for the internal BL616 in the Console 60K and newer Tang20Ks:

1. You mention "Tang Console 60K requires an extern powered USB Hub (or HW modification related to USB-C powering)" - could you clarify both points? Both why this is needed (what the problem is) and what the modification is
2. More generally, I wanted to check whether there's some ordering of operations issue here:

- I have no problem flashing the internal BL616 via BLFlash, but once I've done that no combination of keys seems to be able to bring up a serial interface to make it possible to use either the Gowin IDE or openfpgaloader to flash the required NanoApple2 .fs file.
- Is it then the case that we should flash the .fs first, then the BL616? Or have I missed something?

harbaum, 2025-07-14:
There might actually be a more elegant solution. You can add user jtag endpoints to the fpga. We could thus optionally use jtag for the MCU-FPGA connection.

I am just not convincef that it's worth the effort.

vossstef, 2025-07-14:
> There might actually be a more elegant solution. You can add user jtag endpoints to the fpga. We could thus optionally use jtag for the MCU-FPGA connection.
> 
> I am just not convincef that it's worth the effort.

Think that HW modification attempts are not worth considering risk and effort. usb hub is anyway needed for Gamepad / Joystick. The Keyboard Combo will typically not be sufficient.

vossstef, 2025-07-15:
@HeathenUK did you got your Console programmed and Core working ?

HeathenUK, 2025-07-17:
> [@HeathenUK](https://github.com/HeathenUK) did you got your Console programmed and Core working ?

I'm afraid not. As I say I can successfully flash the internal BL616 easily enough but no matter I try (whether before the BL616 is flashed or not) I can't seem to get openfpgaloader to write to the FPGA itself.

I appreciate this is possibly beyond the scope of the project, but might it be worth putting together an end to end guide as part of the docs for how to take a "stock" Console60K and flash both of its components? It may well be that I'm just not quite registering some alternative way to put the Gowin into download mode, or similar...