MiSTle-Dev/FPGA-Companion #77: Various issues with the ExFat filesystem

Mangerrione, 2025-07-20:
To use HDD-images larger than 4 GB, the ExFat filesystem is required. Which file allocation size is recommended?
I found out 512kb works.

Experimenting with the ExFat filesystem i found various issues...

The NanoMig could not write it's config file onto the SD-Card.

Using a 14 GB HDF containing 3 partitions the 3rd partition did not show up in Amiga OS. The 3rd partition starts beyond the 4GB limit. But everything works fine when run in WinUAE, from the same SD-Card.

Random files on the 2nd partition don't show up on the NanoMig in Amiga OS. Maybe the first partition (300MB) works well, otherwise the system would not boot up. But everything's fine with WinUAE.

Mangerrione, 2025-08-08:
Have built and flashed the latest NanoMig and FPGA-Companion. The issues with ExFat/Large Disks seem unsolved. Amiga OS can not read the 3rd partition and random files on the 2nd partition don't show up. So maybe there is still something wrong in the filesystem handling?

harbaum, 2025-09-20:
Did you retry? The necessary changes of NanoMig required for that weren't IMHO part of the released version when you tested last.

When testing stuff and reporting issues please write which exact release you tested. Or did you build it yourelf from the repository at that time? That should have included the changes.

Mangerrione, 2025-09-20:
I tested it about a month ago. Yes, have buildt Nanomig and FPGA-Companion freshly. Will test again soon and report in detail!

harbaum, 2025-09-21:
I fixed and tested it for the Atari ST and just copied the same code over to the NanoMig.

What does the debug output of the FPGA Companion say? Does it report a big file?

Mangerrione, 2025-09-21:
Sure! Please take the AGS-Image from my Google-Drive: https://drive.google.com/drive/folders/1V6dxx0rj-lLO7XMI7whJWDDMtk67FwY7

Right now the boot process stops because the Work: partition 
can not be found. Just enter "loadwb" and then "endcli" to 
run the Workbench. Start AGS from the workbench menu...

harbaum, 2025-09-21:
Ah ... you are using a bl616 based setup? I never tested that with the large files.

The bl616 does debug output at 2mbit/s and while many USB uarts are supposed to support that it often doesn't work. I this cannot tell if your adapter will work.

Mangerrione, 2025-09-22:
Have tested also Amiga Vision. I has only 9 GB and two partitions. Both partitions show up but random files on the second partition don't exist where they should be. I assume that files after the 4GB barrier are not acessible by the OS...

https://amiga.vision/

Mangerrione, 2025-09-22:
Have set the image to read only in WinUAE.
Write Errors show up immediately...

[image]

Mangerrione, 2025-09-28:
Was able to test the NanoMig core with the latest FPGA-Companion. (RP2040-Zero)
**The large disk support works!**
Strange thing - the NanoMig gets stuck somehow after power on - works right only after a reset...