MiSTle-Dev/FPGA-Companion #102: Core from disk?

Mangerrione, 2025-09-29:
Try to understand how the boot process works with the **on board bl616**.
The **bl616_fpga_partner_nano20k.bin** loads the bitstream from the external flash and transfers it to the FPGA.
When done it hands the bl616 over to **fpga_companion_nano20k.bin**
In order to load a core from disk, the partner firmware would have to be modified?
But that's impossible because it's fused?

Question:
Is there any chance to load a core from the SD-Card on the TN20K?

vossstef, 2025-09-30:
@Mangerrione partner fw is just a black box that takes care of fpga flash programming or in case no usb host attached starts automatically a 2nd stage sw at a predefined location. you can't get rid of it due to encryption and OTP. All Fpga sram load logic need to be part of the companion sw. i think @harbaum might do that for the Pico and own GW2AR 20k board. If base is available then will port over to bl616 environment.

harbaum, 2025-09-30:
On the dev20k board, I have wired the SD card to both, the FPGA and the rp2040. The idea is that the RP2040 initially accesses the SD card directly and boots the FPGA from that. Afterward, the FPGA takes over and all SD card access then goes through the FPGA, even if the rp2040 is accessing it. We'll see if that all works as I hope to ...

This is unfortunately not possible with the TN20k, since not all SD card signals are available on the pin headers.

harbaum, 2025-09-30:
BTW: You'll need a Micro-SD card ...

harbaum, 2025-09-30:
Yes, booting from a USB memory stick should be possible with the Tangs.

I also had one TN20k with a slightly worn out usb-c connector. I did replace it. But that's not trivial ...

Mangerrione, 2025-10-02:
> Yes, booting from a USB memory stick should be possible with the Tangs.

I theory the FPGA-Companion could read bitstream from a USB-Stick and write it to the FPGA on the fly, without writing it to the flash memory?
This could be handy! For testing and development! How hard would it be to implement?