MiSTle-Dev/FPGA-Companion #103: FPGA-Companion & USB-Devices

Mangerrione, 2025-09-30:
A typical user might want to use up to four input devices at the same time, like mouse, keyboard and 2 gamecontrollers or joysticks. 

Have tested different MCU variants (m0s-dock, RP2040-Zero, on board bl616) with various 4-port USB-Hubs. 

From my experience two devices work reliable. When mouse and keyboard are attached with small wireless dongles, it works always. If three or four devices are attached it sometimes works, but often doesn't. 

As this behaviour spreads over different MCU's and various USB-Hub's - it might be a software issue?

Mangerrione, 2025-10-16:
It's just an observation i have made. If i connect keyboard and mouse via small wireless dongles they usually work. If i connect a full size wired keyboard it often doesn't work. If i connect four devices like mouse, kbd and 2 gamepads it works only some times. 
The two DB9 joysticks work always as they should.
The USB-Hubs take their voltage from either the TangNano or from m0s-dock or RP2040-Zero. So i assumed it could be a power issue. Sadly the NEO-Hub can't be powered externally. Will get me powered HUB and try again!

Mangerrione, 2025-10-16:
Now i understand! Also power up issues could be  related to the software. I do hope this issues can be solved. Some pepole will add DB9 ports, others will rely entirely on four USB ports for input devices.

Chandler-Kluser, 2026-09-17:
I have fully built an `.uf2` firmware for RP2040-Zero from Waveshare, but could not get USB to work. Here is my build command:
```bash
$ mkdir build && cd build && cmake -DPICOTOOL_FORCE_FETCH_FROM_GIT=ON -DBOARD=WS2040ZERO -GNinja .. && ninja
```
I also minded what is in `README.md` file to set the board in `CMakeLists.txt` manually:
```cmake
# The BOARD option selects which board to build for
# Regular Pico (W) by default
# set(BOARD PICO CACHE STRING "Board type. Can be PICO, PICO2, WS2040ZERO, SH20KLITE, DEV20K, MSP20K, DEV25K")
set(BOARD WS2040ZERO CACHE STRING "Board type WS2040ZERO")
```
and also patched [```src/class/hid/hid_host.c```](https://github.com/hathach/tinyusb/blob/79445c2386adefb207a76a70d87578b53c3e7922/src/class/hid/hid_host.c#L487) from `tinyusb`:
```c
    tuh_hid_report_received_cb(daddr, idx, (XFER_RESULT_SUCCESS == result)?epbuf->epin:NULL, (uint16_t) xferred_bytes);
```
as described in this [TinyUSB issue #3296](https://github.com/hathach/tinyusb/issues/3296) and [issue #3307](https://github.com/hathach/tinyusb/issues/3307).
I also changed `tusb_config.h` to set up tinyusb host to native rp2040 `D+` and `D-` pins:
```c
// change to 0 if using on-board native micro USB
// change to 1 if using pico-pio-usb as host controller for raspberry rp2040
// #define CFG_TUH_RPI_PIO_USB   1
// to
#define CFG_TUH_RPI_PIO_USB   0
```

My `tinyusb` library is on commit `3af1bec1a9161ee8dec29487831f7ac7ade9e189` --> tag 0.20.0

The microcontroller blinks in a green light but when I connect a gamepad or a keyboard it does not send anything to the FPGA, I also have an M0S Dock with FPGA Companion and it works perfectly in the same FPGA project so the issue is not in the FPGA Side...

I appreciate if someone could help me out! Thanks in advance

harbaum, 2026-09-17:
I haven't built and tested that variant for a long time. I'll take a look.

harbaum, 2026-09-22:
I have built and tested yesterday and it was working as expected.

You can get a serial console with debug output on gpio0/1 as explained at https://github.com/MiSTle-Dev/FPGA-Companion/blob/main/DEBUGGING.md

That may give some more insight?

Chandler-Kluser, 2026-09-23:
> I have built and tested yesterday and it was working as expected.
> 
> You can get a serial console with debug output on gpio0/1 as explained at https://github.com/MiSTle-Dev/FPGA-Companion/blob/main/DEBUGGING.md
> 
> That may give some more insight?

Thank you a lot, Till! I will build it again double checking UART debugging for GPIO0/1 and share the output here!

Chandler-Kluser, 2026-09-25:
@harbaum I have rebuilt again firmware for [rp2040 Zero from Waveshare](https://www.waveshare.com/wiki/RP2040-Zero) with earlier modifications for this specific board and **disabling JTAG** with the following modifications:

`CMakelists.txt` (line 114):
```cmake
# comment that in WS2040ZERO setup to disable JTAG
# target_sources(${PROJECT} PRIVATE ../jtag.c ../gowin.c usb_jtag.c)
```

To avoid linker issues with JTAG funcion calls (also disabled USB device mode):

`mcu_hw.c` (lines 197~200):
```c
// #if (MISTLE_BOARD == 2) || (MISTLE_BOARD == 4) || (MISTLE_BOARD == 5) || (MISTLE_BOARD == 6)
//       tud_task();
//       usb_jtag_poll();
// #endif
```

`mcu_hw.c` (line 2363):
```c
// gowin_upload_core(name);
```

Worked perfectly!!!! Thanks a lot

If you want I can make a pull request if you found useful

harbaum, 2026-09-25:
I am a little confused as I could build it out if the box. Maybe I wasn't on the right branch. I'll try again.

But yes, a pr is appreciated.

harbaum, 2026-09-25:
I should really add a boards mapper file that translates from those ifdef board== something to named defines for specific features. That would make these things much more obvious.

Chandler-Kluser, 2026-09-25:
> I should really add a boards mapper file that translates from those ifdef board== something to named defines for specific features. That would make these things much more obvious.

I can contribute reworking these directives

harbaum, 2026-09-25:
A PR is greatly appreciated. 

Feel free to join our discord:
https://discord.gg/eb2VXCjPH

closed 2026-10-07 by Manger74